Ultimaker/Cura #18306: Error slicing with using supports

npab19, 2024-02-12:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Ender 5 Plus

### Describe your model

tablet mount

### Add your .zip here ⬇️

[CE5P_TabletMountClip.zip](https://github.com/Ultimaker/Cura/files/14249381/CE5P_TabletMountClip.zip)

npab19, 2024-02-12:
Doing more testing, it seems like it happens when useing Tree.

Vandresc, 2024-03-01:
Hey,
We've implemented several fixes in hopes of eliminating most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue persists.
Cheers!

closed 2024-03-01 by Vandresc