Ultimaker/Cura #18313: Fails consistently if "tree" support is selected.

Curtis2012, 2024-02-13:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Ender 3

### Name abnormal settings

Fails consistently if "tree" support is selected. 

### Describe model location

Orientation NA. Fails regardless of model or orientation. 

### Describe your model

Failure occurred on multiple models. 

### Add your .zip here ⬇️

[CE3E3V2_PegBoard6screwdriver.zip](https://github.com/Ultimaker/Cura/files/14261729/CE3E3V2_PegBoard6screwdriver.zip)

SHREYAS1188, 2024-02-13:
Hey @Curtis2012,
I tried your zip file in 
Cura Version  = 5.3.0, 
Printer = Ender 3,
Profile = Standard Quality,
It is getting sliced properly (4 hour 32 minute) , their is no model errors can you please check your Profile/Slicer setting.

Which profile are you using

Ref Img:
[image]

Curtis2012, 2024-02-13:
It slices correctly with "normal" support, but fails with "tree". Did you set to "tree"?

⁣Get BlueMail for Android ​

SHREYAS1188, 2024-02-13:
Yes you the screenshot in the previous comment, the support structure is tree.

Curtis2012, 2024-02-13:
Thanks. Will test with different profiles.

⁣Get BlueMail for Android ​

Vandresc, 2024-03-01:
Hey,
We've implemented several fixes in hopes of eliminating most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue persists.
Cheers!

closed 2024-03-01 by Vandresc