Ultimaker/Cura #18314: full object appears in gcode

Neo-bud, 2024-02-13:
### Cura Version

5.4.0

### Operating System

Windows 10 Pro

### Printer

Monoprice Ultimate 2

### Reproduction steps

I created a gear and tried printing it, but the top layer failed. I'm going to attach my STL and GCODe. If you notice the preview on cura brings the top part of the gear appear in full between layer 186 - 196, and then starts printing it. the whole top cylinder part appears and disappears between 186 -196 which is causing the print to fail.
[STL and Gcode.zip](https://github.com/Ultimaker/Cura/files/14268263/STL.and.Gcode.zip)
  

### Actual results

Everything prints great until layer 186. the issue is after Layer 186 the top Cylinder above the gear doesnt print. I sliced this on Bambu just to see if its a file error but it works fine on Bambu. Just not on Cura. 

### Expected results

Should print the whole file 

### Add your .zip and screenshots here ⬇️

[STL and Gcode.zip](https://github.com/Ultimaker/Cura/files/14268280/STL.and.Gcode.zip)

GregValiant, 2024-02-13:
Thanks for the report.
Here is an ortho view of the model.  You can see there is a gap between the gear and the upper axle portion.  I think Cura is slicing it correctly given the model geometry.  The gap is exactly 1mm.  Since your layer height is 0.1 that means that 10 layers would be missing because there is no model there to slice.  (My layer numbers are different because I didn't use a raft.)
[image]

Do I have this right or am I missing something?

Neo-bud, 2024-02-13:
You must be right I will recheck but not sure how Bambu slicer fixed it
automatically- I will check on the original model again

Neo-bud, 2024-02-13:
How do I enable ortho view on cura ?

GregValiant, 2024-02-14:
"Preferences / Configure Cura / Camera Rendering"
[image]

Neo-bud, 2024-02-18:
> "Preferences / Configure Cura / Camera Rendering"
> image.png (view on web)

Thank you please close the ticket that was probably my design fault

GregValiant, 2024-02-18:
Will do.

closed 2024-02-18 by GregValiant