Ultimaker/Cura #18315: windows 11 is  blocking my running your nightly build

bjbeare, 2024-02-13:
### Cura Version

5.7.0

### Operating System

Windows 11

### Printer

n/a

### Name abnormal settings

n/a

### Describe model location

n/a

### Describe your model

n/a

### Add your .zip here ⬇️

Both the .exe and .msi files are blocked. I am trying to use your 5.7 nightly to verify that you have fixed a slicer problem with tree supports.

gioeleminardi, 2024-02-15:
I have the same problem. The .exe and the .msi for amd64 are blocked in Windows 11 without the option to force run them.

### Workaround
1) Search cmd in Windows 11 and run it with Administrator privileges
2) Copy path of the installer and paste it into the cmd prompt
With this workaround I was able to install the UltiMaker-Cura-5.7.0-alpha.0-win64-X64

HellAholic, 2024-03-01:
Hey we had some issues regarding the signing of windows builds. That issue has been resolved and the builds should include the latest key. You might still see a pop up until we reach critical mass of users.

Cheers!

closed 2024-03-01 by HellAholic