Ultimaker/Cura #18317: Every single file attempted will not slice. Crashes halfway through slice.

Galidah, 2024-02-13:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Ender 5

### Describe your model

All of them. Every model attempted. Will not slice anything.

### Add your .zip here ⬇️

[XYZ 20mm Calibration Cube - 1278865.zip](https://github.com/Ultimaker/Cura/files/14271900/XYZ.20mm.Calibration.Cube.-.1278865.zip)

Galidah, 2024-02-13:
[image]

GregValiant, 2024-02-14:
Thanks for the report.
I'm also running Win 10 and I don't have this problem and this is the first report of this problem with Win10 (there have been problems reported with Win11).

In Cura, use the "Help / Show Configuration Folder" command.  In that folder will be the "Cura.log" file.  Zip that file and post the zip folder here.  It may give a clue as to what is going on with your installation.

Galidah, 2024-02-14:
Thank you Greg. I will grab that info this afternoon for you. 
After fiddling with it quite a bit I have found that it only crashes when using tree supports. If I choose none or normal it will process the slicing perfectly fine. 
I also found that if I go back and forth between clicking the slice button and closing the crash window it will crash 99% of the time. Meaning that randomly it will slice with tree supports but it’s rare.

GregValiant, 2024-02-15:
The "tree support" problem has been reported several times.  There is a "nightly build" of 5.7 available [HERE](https://github.com/Ultimaker/Cura/discussions/18080#discussioncomment-8413837) that you might try.  It isn't an official "release" but it includes some automatic reporting capability when it encounters problems.

Post that log file when you get a chance.

Galidah, 2024-02-16:
[cura.zip](https://github.com/Ultimaker/Cura/files/14303661/cura.zip)

Galidah, 2024-02-16:
[cura.zip](https://github.com/Ultimaker/Cura/files/14303667/cura.zip)

ForsakenRei, 2024-03-02:
Tree support is working randomly now, sometimes it will slice but sometimes not. I will give the nightly build a try.

GregValiant, 2026-10-01:
I'll close this as stale.

closed 2026-10-01 by GregValiant