Ultimaker/Cura #18318: Slicing fails when adding support.

gr8design44, 2024-02-13:
### Cura Version

5.6.0

### Operating System

MacOS 14.3.1

### Printer

Neptune 4 Max

### Name abnormal settings

Cannot narrow down the main issue. It seams, if I change a percentage of infill or type of buid plate adhesion, it will work. Not always. I think it definitly has something to do with support settings though. If I choose a default profile without supports, it slices fine. If I use the same profile in Cura 4.8, it slices fine either way. Only happens in 5.6.

### Describe model location

Center of build plate

### Describe your model

Large section on a 4ft x-wing. Average detail. Nothing too fancy.

### Add your .zip here ⬇️

x-wing body model

gr8design44, 2024-02-13:
So, after trying everything I could think of in the supports area, I thought to create and import a clean file with the current settings, and it worked. So, somewhere along the changing of settings and slicing, the data behind the scenes is getting corrupted or jumbled somehow?

Vandresc, 2024-03-01:
Hey,
We've implemented several fixes in hopes of eliminating most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue persists.
Cheers!

closed 2024-03-01 by Vandresc