Ultimaker/Cura #18319: Gap always added between objects assigned different extruders on the first layer

ansonl, 2024-02-13:
### Cura Version

5.4.0

### Operating System

Windows 10

### Printer

Any printer

### Reproduction steps

1. Import an interlocking dual object 3MF
2. Use any printer and print setting profile
3. Slice it
4. See preview where two objects should be flush against each other with no gap

### Actual results

Cura inserts a gap between objects on first layer.
The second layer has no gap as expected. 

The added gap creates print issues when the two objects are designed to be interlocking and if the second object is small enough, creates holes in the model leading to disconnected features that don't stick well to the buildplate on the first layer. 

### Expected results

First layer should have no gap between objects. 

### Add your .zip and screenshots here ⬇️

[UM2DD_Dual_Extruder_Calibration__Target_B_v2.zip](https://github.com/Ultimaker/Cura/files/14272910/UM2DD_Dual_Extruder_Calibration__Target_B_v2.zip)

Gap between objects on first layer when objects assigned to different extruders
[image]

No gap between objects on first layer when objects are assigned to same extruder
[image]

More examples of the gap creating holes and disconnected features. 
[image]
[image]
[image]

GregValiant, 2024-02-14:
Thanks for the report.
I don't have 5.4 installed anymore.  I checked with 5.2.2 and 5.6.0 and I can't duplicate this.
This is the initial layer with 5.2.2 and  5.6.0 looks the same with no gaps.
[image]

I don't have a printer definition file to match the printer in your project file so I was only able to import the models.  Consequently I don't have your Cura settings.  It would be helpful if you could you post another project file with a different dual extruder printer active.

ansonl, 2024-02-14:
@GregValiant I get the same issue when switching to the standard UM S7 profile so the problem does not need a specific profile to duplicate.

Did you assign the two different parts in the file to different extruders before you sliced? If I assign them to the same color/extruder, there is no gap. 

I'm using 5.4.0 due to the bug in 5.6.0 that does not correctly handle start gcode.

GregValiant, 2024-02-14:
I used my dual extruder Ender and an S5.  This is with an S7 active.
Set your "Initial Layer Horizontal Expansion" to 0.0 for both extruders and see if the problem is resolved.

This is an S7 with Initial Layer HE at 0.0 for both extruders.
[image]

and this is with the Initial Layer HE set to -0.10 for both extruders.  You can see that there is a gap and it only exists on the first layer.
[image]

GregValiant, 2024-02-16:
What I see right now I would call "expected behavior" so there is no bug here.  I do understand what you are describing.

I think this needs to be changed from a bug report to a Feature Request since it is really about modifying the "expected behavior".  Sort of like "Brim on outside only" it would be "Initial Layer Horizontal Expansion on Outside Only".  I don't know if that is possible (multiple meshes on the build plate), but as a _Feature Request_ the Cura team could toss the idea around.

At any rate I'll remove the bug label but I'll leave this open so you can respond here.  If you decide to go with the Feature Request let us know and we can close this one.  I think you could use the same screenshots in the feature request.  They do show what is going on.

ansonl, 2024-02-16:
Yes, I understand, changing this to be a Feature Request would be good. Can you change this existing issue to a feature request or do I need to create an entirely new issue?

GregValiant, 2024-02-16:
I'll change the tag but I can't change the title.  If you would take care of that it might work out without having to duplicate everything.