Ultimaker/Cura #18328: Incorrect window/application class on Linux

anarsoul, 2024-02-14:
### Cura Version

5.6.0

### Operating System

Archlinux

### Printer

Elegoo Neptune 4 Plus

### Reproduction steps

Cura declares incorrect Window/Application class - it contains `com/.https://ultimaker.UltiMaker-Cura`, as a result DE (e.g. Plasma or Gnome) cannot match window to .desktop or appdata file

In KDE Plasma right click on window header, More Actions -> Configure Special Application Settings and observe that Window (Application) class is weird:

[image]

### Actual results

Window (Application) class is incorrect

### Expected results

It should be `com.ultimaker.cura` as in appdata file, .desktop file should be updated accordingly.

### Add your .zip and screenshots here ⬇️

N/A

HellAholic, 2024-02-16:
Hey @anarsoul ,
I'll discuss with the team and will update you.
internal reference CURA-11637

jellespijker, 2024-02-17:
Are you using the AppImage provided by us, or did you install Cura via a system package manager.

I can't reproduce this (see screenshot) using our AppImage (5.7.0-alpha.1) on Manjaro KDE 

[image]

anarsoul, 2024-02-17:
> I can't reproduce this (see screenshot) using our AppImage (5.7.0-alpha.1) on Manjaro KDE

I was using 5.6.0 appimage, but I tried 5.7.0-alpha.1 and I'm seeing the same issue. Are you running it from Wayland or X11 session?