Ultimaker/Cura #18329: Slicing failed with unexpected error

ekimsov, 2024-02-14:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Anycubic Mega X

### Name abnormal settings

Crashes regardless of settings changes

### Describe model location

no

### Describe your model

Watertight and fixed Normals, no avail

### Add your .zip here ⬇️

[Uploading AI3MX_HQ Q2-2 DUN Floor PBR 4x4 F 01 v02.zip…]()

HellAholic, 2024-02-16:
Could you try with this build: https://github.com/Ultimaker/Cura/discussions/18080

I expect that it will still crash, but we've added extra logging so that we can (hopefully) find what is causing this.
Also without a project file it is impossible to help you further.

ekimsov, 2024-02-16:
Thank you for getting back to me on this. When you say "project file", I am not sure which file format you are looking for. I though I had added everything to the attached zip file. 

I tried to install the built you referenced, but my system flags it as highly untrustworthy and prevents me from installing it (windows 11). It does not even give me an option to "install anyway", and even running as admin does not make a difference. This happens both for the exe and the msi download.

I will try and research how to circumvent this installation issue at the weekend and get back to you then.

HellAholic, 2024-02-16:
For the project file, it wasn't properly uploaded I believe. When I tried to download it from the ticket it links me back to the ticket so there is no file available.
We're currently working on resolving the blockage from installation on windows. Keep an eye out on the Discussion page for the alpha and there should be some updates coming soon.
Cheers!