Ultimaker/Cura #18330: Drop Down Support Mesh

KEA-Makerlab, 2024-02-15:
### Cura Version

5.6

### Operating System

Windows 10

### Printer

Ultimaker S5

### Reproduction steps

1. Have a custom mesh made for support. (not breaking the 45 angle rule)
2. Per model setting -> Print as support
3. Select settings (in Print as support dialogbox)
4. Uncheck Drop Down Support Mesh

### Actual results

A support structure is added, straight down to the build plate to the custom support mesh. a support for the support so to speak.

### Expected results

Volume of custom mesh should be printed as support and nothing more (Worked in Cura 4.9)

### Add your .zip and screenshots here ⬇️

[Custom Support Bug.zip](https://github.com/Ultimaker/Cura/files/14294233/Custom.Support.Bug.zip)
[image]
[image]

GregValiant, 2024-02-15:
Thanks for the report.
 I'm getting the same result.  I checked with 4.13.1 and the same thing happens.  I haven't been able to keep support from growing for the custom support even with "generate support" turned off for the project and a support blocker in place on that ear.

The Cura team will take a look.

HellAholic, 2024-02-16:
I'll bring it up with the team to see what can be done.
Internal reference: CURA-11636

WhyIcanBuyMe20atAnyMarket, 2024-08-04:
Hi, I have a similar issue, using normal (non-custom) support with certain model geometry:

Model geometry must be such that there is suspended support (as in: bottom of support is model, top of support is model) with a gap underneath it — the model doesn't go all the way down from the support to the bed.

With default settings, the 'horizontal expansion' minus the 'support x/y distance' is less than the minimum line width, so no drop down support prints where there is a model side under the edge of the support.
However, if the model then slopes away, then there's a support printed, underneath the model, for just the depth of the horizontal expansion above. This isn't blockable with support blockers, I assume because it's considered to be a part of the support above, so that needs to go in one chunk. 
[image]