Ultimaker/Cura #18331: showing 1 smoothing group

DigitalSketch, 2024-02-15:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Creality CR-10

### Reproduction steps

Export OBJ with smoothing groups from 3DS Max, Import OBJ into Cura

### Actual results

The display shows the model with just one smoothing group, making it look odd.

### Expected results

Display the proper smoothing groups in the "prepare" section

### Add your .zip and screenshots here ⬇️

[Cura.zip](https://github.com/Ultimaker/Cura/files/14296002/Cura.zip)

GregValiant, 2024-02-15:
Thanks for the report.
I'm not familiar with the term "Smoothing Group".  What are you referring to?  I see a model with what appears to be 4 protrusions and a separate piece to the side.
In addition, the larger cube seems to have a flipped normal.  There is a red surface pointing up and I get the "not watertight" warning.
Here is the slice.
[image]

DigitalSketch, 2024-02-15:
The piece is just one example, it happens with everything I import now. The model I sent was just part of a project I'm working on and needs to stay together. 

Smoothing groups are a visual thing with 3D applications. In 3D, the smoothing groups tell the program to display a model smoothed with the nearest edge if it has the same smoothing group. So a sphere might look faceted if it doesn't have any smoothing groups, but will look smooth if all the faces share the same smoothing group (see the image). 

In previous versions, the Prepare section would look correct with no smoothing, like it does after you slice it, but now it shows it like the image sent in the original post. 

Hopefully, the attached image will explain it a bit
[image]

GregValiant, 2024-02-15:
The exported model file is a mesh consisting of triangles.  Depending on what CAD software you are using, you might be able to export a model with a higher resolution/greater number of triangles, but when the model gets to the slicer, it just a model made up of less or more triangles.  The "smoothing group" as defined in CAD won't make it into the exported model.  It is something that would not be understood by the export format (STL, OBJ, whatever).
This is the same sphere.  The one on the left was exported as a low resolution model and the one on the right as a high resolution model.
[image]

Here they are in Cura.
[image]

This is your project in MS 3D Builder.  It's just the model.
[image]

fieldOfView, 2024-02-15:
The model has invalid (or at least "strange") normals. Either export the model without normals (eg as an STL), or replace the normals using the Mesh Tools plugin.