Ultimaker/Cura #18333: Angle of Slope Slice

Y13Lab, 2024-02-15:
### Is your feature request related to a problem?

I've been looking for a slicer for a long time that has the function of slicing at an angle.

### Describe the solution you'd like

In the slicer CrealityPrint, in the experimental section
there is a setting Angle of Slope Slice.
When this setting is enabled, the slicer prints at the selected angle.
[image]
In this mode, movement occurs in three coordinates x,y,z.

Please add this functionality to cura slicer.

### Describe alternatives you've considered

Until recently, I thought that there were no alternatives.
But I came across a slicer Crealityprint in which the angled cutting mode is implemented.
I was very happy, but after a lot of trying I realized
that it is poorly implemented and not suitable for full use.

### Affected users and/or printers

This is a feature needed by 3D printing enthusiasts and innovators.

HellAholic, 2024-02-16:
Hey @Y13Lab ,
We had a discussion about this feature a while back. See https://community.ultimaker.com/topic/35805-slice-at-an-angle/
I'll bring it up with the team to see if it's something we can pick up.

HellAholic, 2024-02-20:
Given the complications around this feature, we believe we will not work on this anytime soon and will therefore close this issue. If there are any updates regarding similar functionalities or features from the community developers we will included it in the future.

Cheers!

closed 2024-02-20 by HellAholic