Ultimaker/Cura #18337: Variables in machine start Gcode don't work properly

GeraldMckeown, 2024-02-15:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Ender 3 Max - with MicroSwiss direct drive extruder

### Reproduction steps

After upgrade from Cura 5.5.0, the 
M109 S{material_print_temperture_layer_0}; Wait for the extruder to come to temperature
which is included in the printer's startup GCODE, produces
M109 S0; Wait for the extruder to come to temperature
(which turns the extruder OFF!)
Note that the equivalent command for the print bed (M190) works correctly.

It appears that in 5.5, the commands are ignored, which in my case doesn't cause a problem
[Ender start gcode.txt](https://github.com/Ultimaker/Cura/files/14302215/Ender.start.gcode.txt)

### Actual results

The hotend is turned off, and nothing actually prints

### Expected results

Print should have worked with the hotend set at the print temperature specified.

### Add your .zip and screenshots here ⬇️

I have since downgraded to 5.5. (and removed all temperature references from my start GCODE)
Files print normally.

GregValiant, 2024-02-15:
Thanks for the report.
There are three typos in your startup.
M140 S**{material_bed_temperature_layer)0}**; Start heating the bed  >>>>>should be an underscore, not a "close parenthesis".

M104 S**{material_print_temp}**; Heat nozzle to print temperature >>>>>> should be {material_print_temperature_layer_0}

M109 S{material_print_temperture_layer_0} >>>>>"temperature" is misspelled.

I'll go ahead and remove the bug label and close this.  All of those keywords work for my Ender 3 Pro but they have to be spelled and punctuated correctly or they will be ignored.

closed 2024-02-15 by GregValiant

GeraldMckeown, 2024-02-16:
Thank you!

Gerald