Ultimaker/Cura #18338: Unexpected Error

kooziekustomz, 2024-02-16:
### Cura Version

5.6.0

### Operating System

MacOs Big Sur

### Printer

Anycubic Mega S

### Name abnormal settings

No

### Describe model location

no

### Describe your model

yes

### Add your .zip here ⬇️

[Derailleur pulley wheels (Race+MTB) - 2863096.zip](https://github.com/Ultimaker/Cura/files/14309023/Derailleur.pulley.wheels.Race%2BMTB.-.2863096.zip)

HellAholic, 2024-02-16:
Hey @kooziekustomz,
It is difficult to see the issue without a project file and with so many models.
We're addressing some of the crashes that the community is helping us track. They will be part of the next Cura release, 
If you would like to investigate this issue further, I would recommend sharing a project file with your settings and printer included rather than the STL files.

HellAholic, 2024-03-01:
Hey,
We've implemented several fixes in hopes to eliminate most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue still persists.

Cheers!

closed 2024-03-01 by HellAholic