Ultimaker/Cura #18340: Fail slicing

Alexandrio1, 2024-02-16:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

FlyingBearGhost6

### Describe your model

Tech model

### Add your .zip here ⬇️

[rearbody-experimental-left2.zip](https://github.com/Ultimaker/Cura/files/14310670/rearbody-experimental-left2.zip)

Alexandrio1, 2024-02-16:
использовал древовидные поддержки

HellAholic, 2024-03-01:
Hey,
We've implemented several fixes in hopes to eliminate most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue still persists.

Cheers!

closed 2024-03-01 by HellAholic