Ultimaker/Cura #18345: Cannot slice 25 instances of the file

iu2frl, 2024-02-16:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Neptune 4 Pro

### Name abnormal settings

Right click on the model and multiplied 25 times to fill the bed

### Describe model location

25 instances on the bed

### Describe your model

Single slicing works

### Add your .zip here ⬇️

[Clip_and_Chain_for_Elegoo_Neptune_4_series.zip](https://github.com/Ultimaker/Cura/files/14315065/Clip_and_Chain_for_Elegoo_Neptune_4_series.zip)
[image]

iu2frl, 2024-02-16:
Issue seems to be related with the tree setting in supports, disabling them solves the issue
[image]

HellAholic, 2024-03-01:
Hey,
We've implemented several fixes in hopes to eliminate most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue still persists.

Cheers!

closed 2024-03-01 by HellAholic