Ultimaker/Cura #18346: impossible ad geeetech A20

FabioGAEL, 2024-02-16:
### Cura Version

5.6.0

### Operating System

windows 11

### Printer

geeetech A20

### Reproduction steps

trying to add printer Geeetech A20

### Actual results

program CURA crash and A20 not added

### Expected results

geeetech A20 present in printers

### Add your .zip and screenshots here ⬇️

no screenshot available

GregValiant, 2024-02-17:
Thanks for the report.
It appears that the Geeetech A20 is missing it's extruder file.
In Cura, use the "Help | Show Configuration Folder" command to locate your config folder.
Within that folder is an "extruders" folder.
Unzip this file and put "geeetech_A20_1.def.json" into the "extruders" folder.  That should allow you to install the printer.
[geeetech_A20_1.def.zip](https://github.com/Ultimaker/Cura/files/14318731/geeetech_A20_1.def.zip)

Let us know if that works for you.

FabioGAEL, 2024-02-19:
It works!
Thank you for the help

Fabio Franchini

HellAholic, 2024-02-19:
Thanks @GregValiant & @FabioGAEL 
I restored the file in a PR, should be fixed once that is merged with the new version of Cura.

For reference #18372

closed 2024-02-19 by HellAholic