Ultimaker/Cura #18348: ironman helmet slicing failed

brelyt300, 2024-02-17:
### Cura Version

5.6.0

### Operating System

windows 10 

### Printer

QIDI x plus 

### Name abnormal settings

default 

### Describe model location

centered on build plate 

### Describe your model

stl pre split helmet

### Add your .zip here ⬇️

[Mk 2 Helmet Base.zip](https://github.com/Ultimaker/Cura/files/14317045/Mk.2.Helmet.Base.zip)

GregValiant, 2024-02-17:
Thanks for the report.
I didn't have any trouble slicing the model so I am unable to duplicate the problem.  The model isn't watertight though and might not print correctly.
--> 6 Non-manifold edges
--> 10 Inverted faces
This is with the model repaired in MS 3D Builder.  It looks the same as the un-repaired model.
(0.2 layer height and 0.4 line width.)
[image]

Vandresc, 2024-03-01:
Hey,
We've implemented several fixes in hopes of eliminating most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue persists.
Cheers!

closed 2024-03-01 by Vandresc