Ultimaker/Cura #18353: introduce brick-layering as from CNC Kitchen Video

berberic2, 2024-02-17:
### Is your feature request related to a problem?

Should improve layer adhesion

### Describe the solution you'd like

Please take a look at the excellent video from CNC Kitchen, where he tests what he calls brick layerning.
Effectifly moving every other perimeter-line up ½layer-height.

https://www.youtube.com/watch?v=5hGm6cubFVs

### Describe alternatives you've considered

none

### Affected users and/or printers

Would be an option.

discite, 2024-08-12:
This will introduce a 10% strength, definetly interested in seeing it on the future

berberic2, 2024-11-12:
Take a look at https://www.youtube.com/watch?v=9IdNA_hWiyE  to shed some light at the patent situation.

MaximeGranizo, 2024-11-24:
This would be a very useful feature! Improving layer adhesion is crucial for print quality, and incorporating the brick layering method as demonstrated in the CNC Kitchen video seems like a practical and innovative solution. It would be great to see this as an option for users looking to enhance their prints!

tilmaen, 2025-01-11:
I would totally love this feature.
The european Patent hasn't been filed yet. it seems it is stuck because of a more thorough examination process. 3rd-Party observations have been filed (and acknowledged), also the examiner seemed to have found several issues of "non novelty", Also the bungled up references to the prior Patents were critizized by the EPC officer (as opposed to the USTPO).
The correspondence is pretty interesting [https://register.epo.org/application?number=EP21175830&lng=en&tab=doclist](url)
i have a friend who works there (different department) i might ask him, whether he can tell me something about that patent application.

mahtDFR, 2025-01-23:
Looks like it's been sort of implemented now...
https://www.youtube.com/watch?v=EqRdQOoK5hc

BarsMonster, 2025-01-26:
> The european Patent hasn't been filed yet. it seems it is stuck because of a more thorough examination process. 3rd-Party observations have been filed (and acknowledged), also the examiner seemed to [...]

Apparently, there is exact match with expired Stratasys patent, making all new patents on brick layers invalid. 
https://patents.google.com/patent/US5653925A/en (expired in 2015)
https://www.youtube.com/watch?v=9IdNA_hWiyE

BarsMonster, 2025-02-14:
@Flopalop2 Yes, regardless of patent validity (i.e. is there sufficient prior art) to protect Ultimaker it is important NOT to introduce "Simplified brick layers" feature as a checkbox. Having separate outer wall configuration thickness for odd and even layers, where end user makes a decision how they want to print - is a safer way to go.

Also, my proposed solution does not look exactly like Fig5. if outer surface is on the left, it is proposed to allow difference on thickness on the inner side, which enhances adhesion to infill (which is not covered in the patent).

Picture from @GregValiant  :

[image]

Further changes can be made to increase "legal separation" from the patent: have outermost perimeter thickness identical, and only introduce thickness variability on second perimeter and deeper inside the print (to increase strength without affecting fine surface details).

Flopalop2, 2025-02-18:
That could totally work. Adding configuration to any implementation would allow for looser interpretations of the patentable idea. The alternate extra wall and connect infill would be replaced with the offset layers. Good idea. I'm hoping Ultimaker steps up and allows one of these options (or both). Until then, I'm just trying to keep these threads alive haha.

BarsMonster, 2025-02-18:
For these interested in intesting, in #20189 there is already experimental build working.

bondarchook, 2025-06-02:
New insights into the "brick layers" technique from CNC Kitchen:
https://www.youtube.com/watch?v=dDgA51zdfLc

TLDR: need to increase flow while using the "brick layers" technique for maximum strength.

Flopalop2, 2025-06-02:
Now, having extra flow is still important for closing both horizontal and vertical gaps if you want a really solid part, since perfect flow the nozzle puts out a rounded bead, and round objects don't mesh perfectly so you have to increase the flow and add part size changes to compensate.