Ultimaker/Cura #18359: Print head goes up and down when printing a small area.. Only with version 5.6.0. With version 5.5.0 no problem.

roel8032, 2024-02-18:
### Cura Version

5.6.0.

### Operating System

Windows 11 Pro

### Printer

Anet A6 and Infitary M508

### Reproduction steps

Slice a product ending in small area or pointed.

### Actual results

Print head stops printing, goes about 4 mm higher and continues.
Same product sliced with Cura 5.5.0. does not have the problem.

### Expected results

Not this behavior.

### Add your .zip and screenshots here ⬇️

[switch_press_bosch.zip](https://github.com/Ultimaker/Cura/files/14322331/switch_press_bosch.zip)

GregValiant, 2024-02-18:
Thanks for the report.
Without a project file I have to guess at what your settings might be.
In the "Cooling" settings of Cura see if "Lift Head" is enabled.  The setting is an override for when "Minimum Layer" time causes a conflict with "Minimum Speed" and will cause the nozzle to lift and pause before continuing.
If you search the gcode you may find ";Small layer, adding delay".  That would be added by "Lift Head".
Use the "File | Save Project" command and zip the 3mf file and post the zip folder here.

roel8032, 2024-02-18:
Yes i find ;Small layer, adding delay in the .gcode file.

[switch_press_bosch.zip](https://github.com/Ultimaker/Cura/files/14323723/switch_press_bosch.zip)

GregValiant, 2024-02-18:
Set the setting visibility to "All" and "Lift Head" is at the bottom of the "Cooling" section of settings.  Turn off "Lift Head" and the print should continue without the extra movement.
I'll go ahead and remove the bug label and close this.

closed 2024-02-18 by GregValiant

roel8032, 2024-02-18:
OK, thanks