Ultimaker/Cura #18361: Water Dam

SpielbahNer, 2024-02-18:
### Cura Version

5.5

### Operating System

Windos 11

### Printer

Ender 5 plus

### Name abnormal settings

no

### Describe model location

no

### Describe your model

Sliced on for ender 3S1 and anycubic 

### Add your .zip here ⬇️

[CE5P_WATER DAM_2_350x200_b.zip](https://github.com/Ultimaker/Cura/files/14323249/CE5P_WATER.DAM_2_350x200_b.zip)

GregValiant, 2024-02-18:
Thanks for the report.
It wouldn't slice for me as received.  I rotated the model 90° CW and it sliced.
Someone from the Cura team will take a look.
[image]

SpielbahNer, 2024-02-19:
Hello Greg,

Thanks for the feedback.

I have now also managed to slice the model.

Just like you, I turned it. I cannot determine where the error lies or what the cause is. 

The model fits into the installation space, there were no errors here.

The STL file is also OK. I have already processed this model several times with Curta and printed it on different printers. This was the first attempt with the Ender 5.

 

Regards

Thomas

 

image.png (view on web) <https://github.com/Ultimaker/Cura/assets/64202104/ebe0309b-bb86-4b53-80ed-d102d6e689c9>

HellAholic, 2024-03-01:
Hey,
We've implemented several fixes in hopes to eliminate most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue still persists.

Cheers!

closed 2024-03-01 by HellAholic