Ultimaker/Cura #18364: Sliced model differs from unsliced model.

benclewett, 2024-02-18:
### Cura Version

 5.6.0

### Operating System

Mac 14.2.1 (23C71)

### Printer

Creality v2 Neo

### Name abnormal settings

None

### Describe model location

Same result.

### Describe your model

A fractol landscape with hole set into it.

### Add your .zip here ⬇️

As you can see from the before and after, the original modal has a hole set into it.  After slicing, the hole is filled in.  This seems wrong, and not what I want.
[Archive.zip](https://github.com/Ultimaker/Cura/files/14324563/Archive.zip)

GregValiant, 2024-02-19:
Thanks for the report.
Cura will color a surface red when it requires support.  Red surfaces facing upward obviously don't need support and so that indicates that the surface is "Flipped" inside-out and Cura can't figure out the geometry and where the outside ends and the inside begins.
A lot of the red surfaces are planar on the top of this model.  That makes it look like a "subtraction" was performed and it was botched.
This model needs extensive repair before slicing it.  I'll remove the bug label but leave this open so you can respond here.

This is the model "as received" with the Cura error message informing the user that it won't slice correctly, and showing the flipped surfaces.
[image]

This is the model after being repaired in MS3DBuilder.  No more red surfaces facing upward.
[image]

Here is the repaired model sliced.
[image]

benclewett, 2024-02-19:
Apologies that this is not your bug.

The model is complete, there are no gaps, and was exported from Blender.  I
guess I need to submit a bug to Blender...

Thanks for your time, that is appreciated.

Beb

GregValiant, 2024-02-19:
I don't know Blender but this may just be a problem with the way the subtraction was performed.  There are a lot of flipped surfaces on the flat area in the front of the model as well.
I'll go ahead and close this.

closed 2024-02-19 by GregValiant