Ultimaker/Cura #18365: not been able to slice anything with supports today (not tried to slice with supports in a long while)

Elliehanna, 2024-02-19:
### Cura Version

5.6.0

### Operating System

windows 10

### Printer

Ankermake m5c

### Name abnormal settings

turned on supports (auto tree supports)

### Describe model location

middle of build plate

### Describe your model

t-rex, tried 2 moddles, 1 from loot studio and one from lord of the print

### Add your .zip here ⬇️

[EnricoDino_75mm_NoSupports.zip](https://github.com/Ultimaker/Cura/files/14325490/EnricoDino_75mm_NoSupports.zip)

GregValiant, 2024-02-19:
Thanks for the report.
I don't find any errors in the models and it sliced for me in 5.6.0 so I'm unable to duplicate the problem.
Use the "File | Save Project" command to create a 3mf project file.  It will contain your settings and printer.
Use the "Help | Show Configuration Folder" command and find the Cura.log file.
Zip the log file and the 3mf file and post the zip folder here.
[image]

HellAholic, 2024-03-01:
Hey,
We've implemented several fixes in hopes to eliminate most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue still persists.

Cheers!

closed 2024-03-01 by HellAholic