Ultimaker/Cura #18375: Fusion bot, unable to slice

GeatMeat, 2024-02-19:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Kobra 2

### Name abnormal settings

Bridge enabled and tree supports

### Describe model location

Centered

### Describe your model

ran mesh fix w/ plugin, has been printed before on a similar printer 

### Add your .zip here ⬇️

[CFFFP_Fusion20bot20v120kit20card_2.zip](https://github.com/Ultimaker/Cura/files/14334690/CFFFP_Fusion20bot20v120kit20card_2.zip)

GregValiant, 2024-02-19:
Thanks for the report.
Cura has been struggling under Windows 11 so it might be that bug.  It sliced for me on Win 10 but there are errors in the model. 

Here I've used Mesh Tools "Split model into parts".  After moving the main portion of the model away, this is what's left.
All the red surfaces facing upward are in error.  Those "slivers" are only 0.016mm thick and wouldn't want to print if they are left separate.
[image]

This file has been repaired and all the little pieces have been merged into the main model (using MS 3D Builder).  See if this makes a difference.
[GV_Fusion20bot20v120kit20card_2.zip](https://github.com/Ultimaker/Cura/files/14334951/GV_Fusion20bot20v120kit20card_2.zip)

GeatMeat, 2024-02-20:
Thanks for the fix, I'll have to look into ms builder for the next time I get this issue. I know the few times when this has happened to me I've switched to older versions of cura and that's done the trick, but I just got rid of all my old installs (of course). But I guess It's always best to have a more proper fix.

closed 2024-02-20 by GeatMeat

GregValiant, 2024-02-20:
I've been doing triage here for going on 3 years.  Prior to the "Won't slice" problems of 5.5 and 5.6 probably 85% of the bug reports were actually the fault of bad models, and about 10% do to improper settings.  Most now are the Won't Slice bug but they are all about the same bug.
Always check your models whether they are downloads or something you designed yourself.  That can eliminate a lot of wasted effort.
PrusaSlicer has some kind of deal with AutoDesk to use the NetFabb repair utility.  Cura just has what FieldOfView was able to accomplish with Mesh Tools so Cura has some capability but it doesn't integrate a full blown repair utility.
Microsoft 3D Builder is pretty good, but non-intuitive.  I use it to repair and/or modify model files though.  It's much better than Mesh Mixer.