Ultimaker/Cura #18378: Slicing Failed

JALinker, 2024-02-19:
### Cura Version

5.6.0

### Operating System

WIN 11 23H2 22631.3155

### Printer

Ender-3 V3 SE

### Describe model location

middle various rotations

### Describe your model

first time for this model

### Add your .zip here ⬇️

[CE3V3SE_pantherme.zip](https://github.com/Ultimaker/Cura/files/14337514/CE3V3SE_pantherme.zip)

HellAholic, 2024-02-23:
Thanks for the report!
I checked the model in the latest internal build and it sliced successfully.
[image]

Issue should be resolved in the next Cura release. If you still experience the issue, please feel free to re-open.
Cheers!

closed 2024-02-23 by HellAholic

JALinker, 2024-02-23:
Any way to sign up to be a beta tester?