Ultimaker/Cura #18379: [5.6.0]Error during slicing

270mag, 2024-02-19:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Elegoo Neptune 4 Pro 

### Reproduction steps

I just sliced my part. It didn't give an error code or reason for the failure.

### Actual results

Slicing get about 75% to complete, then I get the error message, "An unexpected error occurred."

### Expected results

a proper slice

### Add your .zip and screenshots here ⬇️

[CFFFP_Both wings.zip](https://github.com/Ultimaker/Cura/files/14337584/CFFFP_Both.wings.zip)
[cura.log](https://github.com/Ultimaker/Cura/files/14337585/cura.log)

HellAholic, 2024-02-23:
This mesh seems faulty.
You can see it in the log as "Tried to add an item to batch without mesh"

Have you tried doing a quick mesh fix with the Meshtools Plugin? 
You can get it here: https://marketplace.ultimaker.com/app/cura/plugins/fieldofview/MeshTools
If that doesn't fix it you can try a more elaborate mesh fixer like https://myminifactory.github.io/Fast-Quadric-Mesh-Simplification/, or https://www.nanodlp.com/repair-3d-stl-file/.
Or you can use the "Make Solid" feature in AutoDesk MeshMixer. https://meshmixer.com/download.html

270mag, 2024-03-09:
Thanks. Got it working.

closed 2024-03-09 by 270mag