Ultimaker/Cura #18384: Slicing failed. Standard settings.

Grandsolution, 2024-02-20:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Anycubic Chiron

### Name abnormal settings

Normal settings

### Describe model location

90 degrees

### Describe your model

Very simple model

### Add your .zip here ⬇️

[Connection test.zip](https://github.com/Ultimaker/Cura/files/14344792/Connection.test.zip)

whitnejo, 2024-02-22:
I also have the same slicing error while slicing a model turned 90 degrees for my Chiron using Cora 5.6.0 but MacOS 14.2.1. Same model slices fine for my Ender3 using the same OS and Cura versions.

HellAholic, 2024-02-23:
Hey, I checked your model with the latest internal build and it sliced without issues.
[image]
The fixes will be included in the next release.

Feel free to reopen if you still experience the issue after the next release.
Thanks for you report.
Cheers!

closed 2024-02-23 by HellAholic