Ultimaker/Cura #18385: Slicing failed when model is horizontal

Neo-bud, 2024-02-20:
### Cura Version

5.7

### Operating System

Win 10 Pro

### Printer

Monoprice Ultimate

### Name abnormal settings

Custom Normal

### Describe model location

Center of plate

### Describe your model

Spacex LAunch tower - model split into 2 

### Add your .zip here ⬇️

[MU_WinfixedSpacex Launch Tower Final.zip](https://github.com/Ultimaker/Cura/files/14346876/MU_WinfixedSpacex.Launch.Tower.Final.zip)

GregValiant, 2024-02-21:
Thanks for the report.
It won't slice with your Monoprice Ultimate active.  I tried a couple of other printers and it will slice with my Ender 3 Pro active, or a UM S5 active, but with an AnyCubic Kobra active it won't slice.

The Cura team will take a look.

HellAholic, 2024-03-01:
Hey,
We've implemented several fixes in hopes to eliminate most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue still persists.
Just double checked on latest internal build and it sliced successfully.

Cheers!

closed 2024-03-01 by HellAholic

Neo-bud, 2024-03-06:
Great to hear