Ultimaker/Cura #18386: Unexpected error - is it a memory issue?

jodain, 2024-02-20:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

AC Kobra

### Name abnormal settings

0.08mm layer height, slices ok at 0.12mm

### Describe model location

Tried rotating 90 degres, but made no difference.

### Describe your model

Not sliced previously, it is watertight.

### Add your .zip here ⬇️

[https://1drv.ms/u/s!AmGfEW76ATajlaEtXX3ZrFRWI2vy4A?e=GHybhx](https://1drv.ms/u/s!AmGfEW76ATajlaEtXX3ZrFRWI2vy4A?e=GHybhx)
I've tried 5.7.0 alpha as well with the same issue

GregValiant, 2024-02-21:
Thanks for the report.
There have been a lot of reports lately of Cura not playing nice with Windows 11.  There have also been a lot of reports in regards to models like this not slicing.  Cura seems to struggle with rough stone or wood floors.  I checked the model and didn't find any errors in it.
I moved the model a little from the middle of the build plate and rotated it 90° CW and it seems to be slice OK.  It is slow because there are 1.2 million surfaces in that model and many are within just a couple of layers.

Someone from the Cura team will take a look.

This took 44.55 minutes to slice.
[image]

HellAholic, 2024-02-21:
Hey,
the fail slice issue has been resolved in a ticket we worked on (tested on internal alpha builds).
For internal reference CURA-11444

GregValiant, 2024-02-21:
@HellAholic I'm still getting the crash with this project and a fresh installation of 5.7alpha.   This is on a Dell laptop running Win 10 Pro.
[image]

HellAholic, 2024-02-21:
Hey @GregValiant, I checked it on build from development branch (hence the internal alpha reference).
Special QA privileges and all :)
Sorry if my comment caused confusion.

GregValiant, 2024-02-21:
Alright.  I thought it was the nightly.

closed 2024-03-01 by Vandresc