Ultimaker/Cura #18391: Slicing Failed a no reason given

1MariaSN, 2024-02-21:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Wasp 2040

### Name abnormal settings

None

### Describe your model

Tolerance Test File

### Add your .zip here ⬇️

I've attached the file I'm having issues with. It works in Prusa Slicer.
[Tollerance Test v2.zip](https://github.com/Ultimaker/Cura/files/14362345/Tollerance.Test.v2.zip)

Vandresc, 2024-02-23:
Hi @1MariaSN , it slices for us in our internal build. The fix will be in the next release, but again I didn't have access to your printer. so I used one of our printers (Um S5) for testing.

closed 2024-02-23 by Vandresc

1MariaSN, 2024-02-23:
> You don't often get email from ***@***.******@***.***>. Learn why this is important<https://aka.ms/LearnAboutSenderIdentification>
> CAUTION: This e-mail originated outside the University of Southampton.

Thanks!
Would you happen to know when it would be released next?

Many thanks,

Maria

Vandresc, 2024-02-23:
We try to release every 3 months.
And a little birdy told me that a release is coming up, so you hopefully don't have to wait that long.
Thanks again for reporting! Together we can make Cura better.