Ultimaker/Cura #18392: long slicing process then failed

hondaman2487, 2024-02-21:
### Cura Version

5.5.0

### Operating System

windows 11

### Printer

Anycubic Kobra Plus

### Name abnormal settings

no

### Describe model location

all areas

### Describe your model

sliced once before with another model but slicer has been taking a long time on all models to slice

### Add your .zip here ⬇️

[Uploading sponge holder.zip…]()

HellAholic, 2024-02-23:
Hey @hondaman2487 , we addressed few tickets that resolved the long slice time, it will be included in the next release.
Unfortunately your project file was not uploaded so we cannot use that to test.

HellAholic, 2024-03-01:
Hey,
We've implemented several fixes in hopes to eliminate most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue still persists.

Cheers!

closed 2024-03-01 by HellAholic