Ultimaker/Cura #18397: Missing lines causing others to be printed mid-air

ThomasRahm, 2024-02-22:
### Cura Version

5.7.0-alpha.0

### Operating System

Windows 10

### Printer

Ultimaker 2+

### Reproduction steps

Slice the model. 

### Actual results

[image]
Issue can be reduced (but not fully avoided) by increasing _Wall Transitioning Threshold Angle_ which suggests to me that it is Arachne related.

### Expected results

Lines should be generated for areas that are larger than areas further up for which lines are generated.

### Add your .zip and screenshots here ⬇️

[HobgoblinBug.zip](https://github.com/Ultimaker/Cura/files/14370002/HobgoblinBug.zip)

Just reporting the bug as i noticed it while testing something else. This model is my preferred test model, as it gives plenty opportunity for problems. I do not need help to find a workaround.

HellAholic, 2024-04-08:
Added a ticket for investigation. Internal reference CURA-11826

We had a similar-ish behavior reported for CURA-11591 but the source could be different.

MariMakes, 2024-04-09:
👋  I recognize this report. This seems to be a duplicate of https://github.com/Ultimaker/Cura/issues/12817
I know the linked issue is old, but we discussed the team and reprioritized it. It's now scheduled to be picked soon!

I'll be closing this issue as a duplicate but you can follow the progress here: https://github.com/Ultimaker/Cura/issues/12817

closed 2024-04-09 by MariMakes

casperlamboo, 2024-04-10:
Hi @ThomasRahm,

Yes, very annoying issue indeed. One I have encountered a few times as well. Although the [arachne](https://github.com/Ultimaker/CuraEngine/pull/1210) engine handles narrow walls well, it doesn't handle small areas (dot's circles) too great.

There are two things at play here that both make sense on their own, but simultaneously produce some weird results.

In Arachne lines are generated to be parallel to a "central edge" of a shape. For some shapes (triangles, octagons, circles) there is no such "middle line" but instead there is a "middle dot". 3D printers have trouble printing "dots" as you might imagine, so we print a small circle instead. See https://github.com/Ultimaker/CuraEngine/blob/main/src/SkeletalTrapezoidation.cpp#L2194-L2215.

The second factor at play here is logic that removes too small to print lines; the logic here is that any path that is smaller then it's width can't be printed so such paths are removed.
https://github.com/Ultimaker/CuraEngine/blob/8b1757ec7e7906b138952da2db50934749768469/src/WallToolPaths.cpp#L277-L297

In your model Arachne hovers on the border of adding a center dot and a center line. So on some layers a small circle is created, and on some layers a very short line. The short lines are removed, and the circles have plenty length so they remain.

I have created a small video that highlights this nicely, all "removed" lines are colored red. Note the transition between "cricles" and "lines"

https://github.com/Ultimaker/Cura/assets/6638028/667db787-8295-4cbb-850e-5937a42f7280