Ultimaker/Cura #18400: Creality Ender-3 V3 SE start G-code BUG

zanghao912, 2024-02-22:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Creality Ender-3 V3 SE

### Reproduction steps

There is an error in this field in the start G-code
M109 S[material_print_temperature_layer_0]

### Actual results

This value is not available after slicing with the slicing software.

### Expected results

It should be fixed to
M109 S{material_print_temperature_layer_0}

### Add your .zip and screenshots here ⬇️

not

GregValiant, 2024-02-22:
Thanks for the report.
@HellAholic - will you be putting in a PR to change the brackets?

zanghao912, 2024-02-23:
Yes, I've committed.[image]

HellAholic, 2024-02-23:
Found the existing PR that fixes this issue https://github.com/Ultimaker/Cura/pull/17781
Thanks for the report.

Cheers!

closed 2024-02-23 by HellAholic