Ultimaker/Cura #18401: Improving Cura with a new Plugin, could call it >Line-Intro< a line from the current (start-) position to Brim/Skirt/Raft/Object

waldab, 2024-02-22:
### Is your feature request related to a problem?

I have a suggestion for improving Cura with a new Plugin, could call it >Line-Intro< a line from the current position to Brim

Problem:
In my start code I entered 2 straight lines, each 30cm long, on the front of the print bed for nozzle cleaning. The slightly raised nozzle then moves without extruding to print the brim (or skirt or raft or object). It happens that the first millimeters of the brim/raft/skirt/object do not adhere correctly and are taken away by the nozzle and the first layer becomes unclean.

My solution:
After the 2 cleaning lines from the start code, I print a line directly to the beginning of the brim/raft/skirt/object, so that the first layer adheres from the start of the Brim. I call this line >INTRO-Line<.
Would anyone find this helpful and can implement this option in a Cura or in a Plugin?

[image]

### Describe the solution you'd like

After the 2 cleaning lines from the start code, I print a line directly to the beginning of the brim/raft/skirt/object, so that the first layer adheres from the start of the Brim. I call this line >INTRO-Line<.
Would anyone find this helpful and can implement this option in a Cura or in a Plugin?

### Describe alternatives you've considered

a Plugin or a post processing Script

### Affected users and/or printers

any and all printers

GregValiant, 2024-02-22:
If the "Layer Start X" and "Layer Start Y" are somewhere across the build plate then you would be extruding across the print footprint.  That would leave a mark on the bottom of the print.
[image]

So this is up to the user and whatever settings they have in Cura.
You can use a "Search and Replace" to make adjustments at the start of Layer 0.

Search:  ;LAYER:0
Replace: ;LAYER:0\nG92 E-9.5

My retract distance is 6.5 so the above addition would provide an additional 3mm of prime at the start of the skirt.  In my case that makes up for the 2mm retraction at the end of my purge lines and provides an additional 1mm to make up for oozing.

waldab, 2024-02-22:
I am aware of the problem with the crossing line, but when using a brim or skirt, Cura always starts the printing process at the front left, i.e. at the shortest distance from the origin (as long as the origin x=0/y=0 is at the front left). This means there is no intersection. The plugin/script would have to be designed in such a way that it also takes all other cases into account and avoids intersections.

GregValiant, 2024-02-26:
_"...but when using a brim or skirt, Cura always starts the printing process at the front left,"_
You can see in my screen shot that isn't true.  The start of the skirt is at the right top which is my case is a Layer Start X=230 and Layer Start Y=230.

waldab, 2024-02-26:
Yes, I was a bit hasty with my statement. If you print multiple parts, the order in which you place the parts influences the starting point. But all of these are details that a postprocessing script could handle. I have already received several opinions that suggest better printing parameters and thus deny the need for such a script. I have now programmed a Windows app that accesses the USB stick, focuses on the latest sliced ​​file, manipulates the G-code and stores it on the USB stick with a prefix. The Windows app works for printing individual parts. Maybe I'll look into Phyton and how to integrate scrips into CURA.

GregValiant, 2024-02-27:
You can use existing post processing scripts as examples.
A definite advantage to a true post-processor (as opposed to post-processing with a separate app) is that any needed Cura settings can be pulled from Cura on the fly.