Ultimaker/Cura #18404: Cura Slicing Haunted House 120%

rockrattled, 2024-02-22:
### Cura Version

5.3.0

### Operating System

Mac OS Sonoma

### Printer

Ender 3

### Describe model location

Rotating works

### Describe your model

I have sliced similar models before

### Add your .zip here ⬇️

Haunted House

Vandresc, 2024-02-23:
Hi @rockrattled , could you try in Cura 5.6? Or provide us with a project file so we can try to reproduce the issue?  Thank you!

HellAholic, 2024-03-01:
Hey,
We've implemented several fixes in hopes to eliminate most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue still persists.

Cheers!