Ultimaker/Cura #18405: Slicing Failed

benclewett, 2024-02-22:
### Cura Version

5.6.0

### Operating System

Mac 14.2.1 (23C71)

### Printer

Creality v2 Neo

### Describe your model

I have slices similar models from the same engine and printed them successfully.  This model is complete and correct.

### Add your .zip here ⬇️

https://we.tl/t-LcwStCB1mc

Vandresc, 2024-02-23:
Hi @benclewett thank you for the report, it was sliced for me in our internal build, so the fix will be in the next release.
 If still an issue please re-open.

closed 2024-02-23 by Vandresc

Vandresc, 2024-02-23:
We try to release every 3 months.
And a little birdy told me that a release is coming up, so you hopefully don't have to wait that long.
Thanks again for reporting! Together we can make Cura better.

benclewett, 2024-02-23:
I’m looking forward to it…..

Thanks for excellent support.