Ultimaker/Cura #18406: Slicing failed - Cura 5.7.0-alpha+ac5afc

csterlin, 2024-02-22:
### Cura Version

5.7.0-alpha.1+ac5afc

### Operating System

Windows 11 Home, 22H2, 22621.1778, Intel(R) Iris(R) Xe Graphics

### Printer

Ender 3

### Name abnormal settings

Print thin walls - Slices okay with this unchecked or with different horizontal expansion setting either higher or lower by 0.1mm

### Describe model location

No, moving it or rotating it made no difference

### Describe your model

Misc parts for a stormtrooper helmet

### Add your .zip here ⬇️

[helmet-black-parts.zip](https://github.com/Ultimaker/Cura/files/14380330/helmet-black-parts.zip)
[cura.zip](https://github.com/Ultimaker/Cura/files/14380355/cura.zip)

Vandresc, 2024-02-23:
Hi @csterlin , thank you for your feedback! I can indeed reproduce this as well, I will bring this up for further discussion with the devs.

HellAholic, 2024-03-01:
Hey,
We've implemented several fixes in hopes to eliminate most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue still persists.
Side note: Just double checked your project on the latest internal build, and no more slice failed, so fingers crossed :)

Cheers!

closed 2024-03-01 by HellAholic