Ultimaker/Cura #18407: 5.6.0 Initial Layer Infill Speed not affected by initial layer speeds.

opjose, 2024-02-23:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

CR-10

### Reproduction steps

On multiple circular (cylindrical) models I set the initial layer print and travel speed to a very slow 12mm/s. The Inner wall and Infill then prints at the normal speed assigned to all other layers. With previous version the infill and inner wall would never exceed the layer print speed.

### Actual results

Inner wall prints at 60mm (what I have as the normal layer speed)

### Expected results

Inner walls and Infill should print no faster than initial layer speed.

### Add your .zip and screenshots here ⬇️

[image]

opjose, 2024-02-23:
This is visible in the Gcode...

;TYPE:SKIRT
G1 F600 Z0.3
G1 F2400 E0
G1 F720 X125.394 Y136.377 E0.03409
...
;MESH:Horizontal_Expansion_Thread_M10X1.5-10mm.stl (OK up to here)
G0 F720 X131.73 Y145.244
G0 X132.516 Y145.99
;TYPE:WALL-INNER (Note F3600 below!)
G1 F2400 E38.69056
G1 F3600 X140.723 Y159.181 E38.71446
G1 X141.217 Y159.063 E38.74081

Asterchades, 2024-02-23:
There is currently a bug with the Initial Layer Speeds in some circumstances, and has been since 5.5. The two most likely sources of the interference are `Top Surface Skin Layers` and `Print Thin Walls` - setting the former to zero (0) or enabling the latter should fix it, though some users have found it necessary to do both.

Some users also report that the (mis)behaviour has been fixed in 5.7, which is currently in alpha.

Duplicate of #17210.

closed 2024-02-23 by Asterchades