Ultimaker/Cura #18416: Nozzle temperature not set

niccolorogai, 2024-02-24:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Creality Ender 3 V3 SE

### Reproduction steps

I set the printing temperature to 200 and slice.

### Actual results

When i open the generated gcode on my printer the nozzle target temperature stay 0 instead of being set to 200. 
Then i need to manually set the nozzle temp to 200 using the settings interface on the printer.

### Expected results

It should set the nozzle target temparature to 200 

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2024-02-26:
Thanks for the report.
There is a typographical error in your StartUp Gcode.
Go to "Manage Printers" and then "Machine Settings".  The StartUp Gcode is in the lower left textbox.
Find this line:
M109 S[material_print_temperature_layer_0]
Prusaslicer uses square brackets but Cura uses curly brackets.
Change that line to curly brackets as:
M109 S{material_print_temperature_layer_0}

@HellAholic here is another for your attention.

HellAholic, 2024-02-26:
Thanks for the report.
PR that fixes this issue https://github.com/Ultimaker/Cura/pull/17781
Issue should be resolved in the next Cura release.

Cheers!

closed 2024-02-26 by HellAholic