Ultimaker/Cura #18418: Fehlerhafter g-code

Toni4Scout, 2024-02-24:
### Cura Version

5.6.0

### Operating System

Windows 11

### Printer

Elegoo Neptune 4Pro

### Name abnormal settings

keine Eistellung verändert

### Describe model location

Modell ist auf dem Mittelpunkt der Druckplatte platziert

### Describe your model

das Modell wird richtig sliced, im G-Code liegt das Problem. Bei der Ansicht auf das Druckverfahren sieht man den Fehler. In der Ersten Druckschicht sollte der Rote Lange Streifen nicht auftauchen

### Add your .zip here ⬇️

[image]

GregValiant, 2024-02-26:
Thanks for the report.
The Neptune 4 isn't included with 5.6.0 (it will be in 5.7).
What are you using for a definition file?  Those look like purge lines that might be in the StartUp gcode (like Creality definitions).

GregValiant, 2026-09-10:
I'll go ahead and close this.

closed 2026-09-10 by GregValiant