Ultimaker/Cura #18421: Reload show different behaviors

UBaer21, 2024-02-24:
### Cura Version

5.6.0

### Operating System

win10

### Printer

Theseus

### Reproduction steps

1. Load an Object
2. multiply the Object
3. make a change to the object and export/save
4. Reload All Models via context menu

### Actual results

[image]
This is one model multiplied 2×
Only the first is reloaded/updated but not the multiplied instances.
If the model is opened a second time this is also updated.

If You click the "File has been modified - want to reload" dialog  then All instances are updated. But that dialog sometimes vanishes or Cura lost track of updating the file and didn't show the dialog.

### Expected results

All models are reloaded and updated.
And that the pop up dialog reload button has the same result as the context menu reload.

### Add your .zip and screenshots here ⬇️

[image]

Why is this a mandatory field to fill out?

qwertymodo, 2024-04-27:
I can confirm this is happening to me as well.  Without having looked at the code, my initial suspicion is that when objects are multiplied, it's modifying the object ID of the multiplied objects in a way that breaks the link back to the original .stl file, so when you attempt to reload all, those copies are trying to reload from a file that doesn't exist.

MariMakes, 2024-11-19:
👋 Quick update on our side.
This is resolved in the 5.9 release 🎉 , you can download the Cura version with the fix here: [https://github.com/Ultimaker/Cura/releases/tag/5.9.0](https://github.com/Ultimaker/Cura/releases/tag/5.9.0)

I'll close this issue since it's resolved.
Thanks again, and please let us know if you run into any other issues 💪

closed 2024-11-19 by MariMakes