Ultimaker/Cura #18424: slicing failed

galofer100166, 2024-02-25:
### Cura Version

5.4.0

### Operating System

windows 11

### Printer

Artillery Hornet

### Name abnormal settings

mi modelo se corta en Orca slicing pero no en ultimaker, la configuracion es normal con respecto a otras que no me dieron problemas

### Describe model location

no

### Describe your model

si, en Orca

### Add your .zip here ⬇️

[Renault_12_-_Buena.zip](https://github.com/Ultimaker/Cura/files/14395057/Renault_12_-_Buena.zip)

Vandresc, 2024-03-01:
Hey,
We've implemented several fixes in hopes of eliminating most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue persists.
Cheers!

closed 2024-03-01 by Vandresc

galofer100166, 2024-03-01:
Ok tanks so much!