Ultimaker/Cura #18426: slicing failed

galofer100166, 2024-02-25:
### Cura Version

5.6.0

### Operating System

windows 10

### Printer

Artillery Hornet

### Name abnormal settings

mi modelo se corta en Orca slicing pero no en ultimaker, la configuracion es normal con respecto a otras que no me dieron problemas

### Describe model location

45 degrees and nothing, plane and nothing...

### Describe your model

yes...from orca

### Add your .zip here ⬇️

[Renault_12_-_Buena.zip](https://github.com/Ultimaker/Cura/files/14395097/Renault_12_-_Buena.zip)

HellAholic, 2024-03-01:
Hey,
We've implemented several fixes in hopes to eliminate most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue still persists.

Cheers!

closed 2024-03-01 by HellAholic

galofer100166, 2024-03-01:
Gracias