Ultimaker/Cura #18433: unable to build .msi setup

Jay22K, 2024-02-25:
### Cura Version

5.4.0

### Operating System

win10

### Printer

no printer

### Reproduction steps

`python  create_windows_msi.py  G:\Projects\freelance_projects\new_cura\Cura G:\Projects\freelance_projects\new_cura\Cura\dist 'UltiMaker-Cura-5.1.0-beta-Windows-X64.msi' 'UltiMaker Cura'`

hit this command and i got error.

### Actual results

this is error i got after i hit above command:

Windows Installer XML Toolset Toolset Harvester version 3.11.2.4516
Copyright (c) .NET Foundation and contributors. All rights reserved.

Windows Installer XML Toolset Compiler version 3.11.2.4516
Copyright (c) .NET Foundation and contributors. All rights reserved.

UltiMaker-Cura.wxs
G:\Projects\freelance_projects\new_cura\Cura\packaging\msi\UltiMaker-Cura.wxs(3) : error CNDL0108 : The Product/@Version attribute's value, 'None.None.None', 
is not a valid version.  Legal version values should look like 'x.x.x.x' where x is an integer from 0 to 65534.
G:\Projects\freelance_projects\new_cura\Cura\packaging\msi\UltiMaker-Cura.wxs(3) : error CNDL0010 : The Product/@Version attribute was not found; it is requir
ed.
HeatFile.wxs
light.exe : error LGHT0103 : The system cannot find the file 'G:\Projects\freelance_projects\new_cura\Cura\packaging\msi\build_msi\UltiMaker-Cura.wixobj' with
 type 'Source'.
PS G:\Projects\freelance_projects\new_cura\Cura\packaging\msi> 

### Expected results

as given i got .msi setup file

### Add your .zip and screenshots here ⬇️

how can i build msi setup file

varshad5, 2024-03-19:
i had the same issue

GregValiant, 2026-10-01:
I'll close this as stale.

closed 2026-10-01 by GregValiant