Ultimaker/Cura #18434: Ender 5 issue with slicing

syfygi, 2024-02-25:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Ender 5

### Name abnormal settings

infill-20%

### Describe model location

yes, If I rotate 90 degrees it slices

### Describe your model

it sliced properly at 100% infill

### Add your .zip here ⬇️

[CE5_Pal-Sphere_-_Simple_Half_B.zip](https://github.com/Ultimaker/Cura/files/14397775/CE5_Pal-Sphere_-_Simple_Half_B.zip)

GregValiant, 2024-02-26:
Thanks for the report.
This sliced for me "as received" in both 5.6.0 and in 5.7alpha.  So I am unable to duplicate the problem.
Moving and rotating the model, I'm not able to get it to "not slice".

HellAholic, 2024-03-01:
Hey,
We've implemented several fixes in hopes to eliminate most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue still persists.

Cheers!

closed 2024-03-01 by HellAholic