Ultimaker/Cura #18435: Slicing error

CurlyShiny, 2024-02-25:
### Cura Version

5.7.0-alpha.0

### Operating System

Windows11

### Printer

Ultimaker 2+

### Describe your model

The model slices when a normal size. When i double the size it gives me the error.

### Add your .zip here ⬇️

[DurgunForearmUpper.zip](https://github.com/Ultimaker/Cura/files/14397818/DurgunForearmUpper.zip)

GregValiant, 2024-02-26:
Thanks for the report.
I can't duplicate this so we will need a project file ("File | Save Project").  Zip it and post the zip folder here.

This is 100% thru 400%.
[image]

HellAholic, 2024-03-01:
Hey,
We've implemented several fixes in hopes to eliminate most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue still persists.

Cheers!

closed 2024-03-01 by HellAholic