Ultimaker/Cura #18436: Gradient from top layer of preview, to visualize the slices better.

fatihG, 2024-02-25:
### Is your feature request related to a problem?

Sometimes it gets difficult to distinguish the separate layers from each other, especially if the slices only have little variation between them. 

### Describe the solution you'd like

The solution I am suggesting, to be able to visualize the slicing better, is to have a gradient starting from the topmost visible layer and getting darker as it goes down the layer.  
This way you will be able to visiually set apart the layers as they will not blend together and have slightly different shades between them. 

This could simply be a static amount of slices the gradient gets applied to or a setting the user could set to apply the gradient tot he amount of layer they want. 
If it is a hardcoded amount I think at least 3 slices would be good. 

### Describe alternatives you've considered

An alternative solution could be to make the top most slice that is being displayed to simply be brighter than the underlying layers. 

### Affected users and/or printers

Anyone that has to inspect the results of the slicer. 

### Additional information & file uploads

[image]

GregValiant, 2024-02-26:
The Preview Slider has two white circles.  The one at the top will lower the layer down to the "layer of interest".  The bottom circle can be slid upward to hide the layers below the "layer of interest".
By sliding the bottom circle up and the top circle down only layer 26 is visible here.
[image]

Is this the sort of thing you are asking for?

fatihG, 2024-02-28:
I am aware of this functionality.

My suggestion was more so to be able to inspect the relation of the slices with each other.
If I want to see how, in the screenshot posted previously, the support structure is generated and how much tolerance there is between the slices it is a bit hard to make out. 

A small step in brightness between the topmost slices and a few slices below it would make it easier to visualize.