Ultimaker/Cura #18440: Error path in slicing

constantinsta, 2024-02-26:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Creality CR-10S Pro V2

### Reproduction steps

I am slicing the object.
And I press preview. 

### Actual results

After the first printed level, the printer head is moving to a specific point, after each line is printed.

### Expected results

A working g-code file without bugs

### Add your .zip and screenshots here ⬇️

[Box.zip](https://github.com/Ultimaker/Cura/files/14402471/Box.zip)
[image]
[image]
[image]
[image]

GregValiant, 2024-02-26:
Thanks for the report.
I agree, that doesn't look right.  It might be a problem with the combing mode "Not in skin".  In the "Travel" section, changing the combing mode to "All" appears to make the problem go away.

The Cura team will take a look.

HellAholic, 2024-03-01:
Thanks for the report. 
We have a ticket on the backlog related to the combin not in skin.
Internal reference: CURA-10308

Cheers!

closed 2024-03-12 by constantinsta