Ultimaker/Cura #18441: Slicing failure

Maskery, 2024-02-26:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Ender 3

### Name abnormal settings

N/A

### Describe model location

Some times it does but not often

### Describe your model

It happens with lots of models. Some I've sliced before some I havent. 

### Add your .zip here ⬇️

na

oskarmay, 2024-02-26:
The same happens to me, a lot lately.
I observe that it sometimes happens when I select supports and sometimes when I select adhesion.
My quick solution for this is to try a couple of times with slicing in different position on the bed (rotation + object move).

I print on Ender 3 v3 SE and slice using cura 5.6.0 on Windows 11.

oskarmay, 2024-02-26:
For future debbuging I add here a zip file with a model that causes error when I try to slice it without any change in position and my current slicer profile.

[model_causing_slicer_error_plus_slicer_profile.zip](https://github.com/Ultimaker/Cura/files/14411209/model_causing_slicer_error_plus_slicer_profile.zip)

HellAholic, 2024-03-01:
Hey,
We've implemented several fixes in hopes to eliminate most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue still persists.

Cheers!

closed 2024-03-01 by HellAholic