Ultimaker/Cura #18442: erro

HenrikStenlund, 2024-02-26:
### Cura Version

5.6.0

### Operating System

win10 64

### Printer

S5

### Name abnormal settings

none

### Describe your model

no

### Add your .zip here ⬇️

[UMS5_spintrap.zip](https://github.com/Ultimaker/Cura/files/14403177/UMS5_spintrap.zip)

HenrikStenlund, 2024-02-26:
parts were added to be printed and then a crash

HellAholic, 2024-03-01:
Hey,
We've implemented several fixes in hopes to eliminate most of the slicing errors. These fixes will be included in the next Cura release that will be available soon.
Please retry with the upcoming beta and feel free to re-open if the issue still persists.

Cheers!

closed 2024-03-01 by HellAholic